epiclabs-uc/esphome-modbus-server #1: flow control pin

NdS-Research-Facilities, 2023-01-17:
Hi, I have a MAX485 board as found here: https://hobbycomponents.com/wired-wireless/663-max485-rs485-transceiver-module

Would it be possible to add a flow control pin, so I can drive this board properly ?

wspeth, 2023-03-04:
> Hi, I have a MAX485 board as found here: https://hobbycomponents.com/wired-wireless/663-max485-rs485-transceiver-module
> 
> Would it be possible to add a flow control pin, so I can drive this board properly ?
@NdS-Research-Facilities: having the same idea/request. I would say, flow control is a must when modbus communicates half duplex. Did you find a solution?

aj-nagrom, 2023-04-05:
I also need a flow control pin functionality like the Modbus Component has: 
https://esphome.io/components/modbus.html

If someone can please provide an update for this that would be amazing!

That-Dude, 2023-04-05:
Untested but this repo has made the flow control pin

https://github.com/arpiecodes/esphome-modbus-server

aj-nagrom, 2023-04-06:
> Untested but this repo has made the flow control pin
> 
> https://github.com/arpiecodes/esphome-modbus-server

Thanks man! I have copied some of the logic and changed a few things so that you can pass GPIO pins in the config rather than an integer number for the pins.

Here is my forked repo:
https://github.com/aj-nagrom/esphome-modbus-server

I have tested it with my setup and I now have modbus 2-way communication! 

I needed to connect both the re and de pins to an output for it to work correctly.

That-Dude, 2023-04-06:
> Thanks man! I have copied some of the logic and changed a few things so that you can pass GPIO pins in the config rather than an integer number for the pins.
>
> Here is my forked repo: [...]

Tested and working!

If you ever add input register support let me know - I'm going to have a try but I'm an awful dev.

timota, 2023-04-08:
Hi guys, 

can you provide ESPhome example code how to connect to [/aj-nagrom/esphome-modbus-server](https://github.com/aj-nagrom/esphome-modbus-server), eg: read sensor var or set switch state.

Thanks in advance

aj-nagrom, 2023-04-12:
> Hi guys,
> 
> can you provide ESPhome example code how to connect to [/aj-nagrom/esphome-modbus-server](https://github.com/aj-nagrom/esphome-modbus-server), eg: read sensor var or set switch state.
> 
> Thanks in advance

This is my code. It doesn't update a switch directly, I called a script that does the update.

There is an example for how to read a sensor at the bottom as well.

```
uart:
  id: modbus_uart
  tx_pin: GPIO17 #TX
  rx_pin: GPIO16 #RX
  baud_rate: 9600
  data_bits: 8
  stop_bits: 1
  parity: NONE
  debug: # Minimal configuration example, logs hex strings by default

modbus_server:
  - id: modbus_server1
    uart_id: modbus_uart
    address: 1 # slave address
    de_pin: GPIO14 #14
    re_pin: GPIO32 #32
    holding_registers:

      - start_address: 1 # LED Color Mode
        default: 0
        number: 1
        on_write: |
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGD("ON_WRITE", "LED Color Mode: address=%d, value=%d", address, value);
          
          // dont let values out of range update the number state - default to 0
          if (value < 0 || value > 7) {
            value = 0;
            ESP_LOGW("ON_WRITE", "LED Color Mode: value is out of range, setting to default=%d", value);
          }
          
          // This method works for directly updating the number value
          //auto call = id(led_color_mode).make_call();
          //call.set_value(float(value));
          //call.perform();

          // This method works via executing a script to update the number value
          id(set_color_mode)->execute(float(value));
          
          ESP_LOGD("ON_WRITE", "LED Color Mode: led_color_mode = %d", value);
          
          return value;
        on_read: |
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGD("ON_READ", "LED Color Mode: address=%d, value=%d", address, value);
          
          int mode = (int)id(led_color_mode).state;
          
          ESP_LOGD("ON_READ", "LED Color Mode: led_color_mode = %d", mode);
          
          return mode;

      - start_address: 2 # LED Flash Mode
        default: 0
        number: 1
        on_write: |
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGD("ON_WRITE", "LED Flash Mode: address=%d, value=%d", address, value);
          
          // dont let values out of range update the number state - default to 0
          if (value < 0 || value > 2) {
            value = 0;
            ESP_LOGW("ON_WRITE", "LED Flash Mode: value is out of range, setting to default=%d", value);
          }
          
          id(set_flash_mode)->execute(float(value));
          
          ESP_LOGD("ON_WRITE", "LED Flash Mode: led_flash_mode = %d", value);
          
          return value;

        on_read: |
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGD("ON_READ", "LED Flash Mode: address=%d, value=%d", address, value);
          
          int mode = (int)id(led_flash_mode).state;
          
          ESP_LOGD("ON_READ", "LED Flash Mode: led_flash_mode = %d", mode);
          
          return mode;

      - start_address: 3 # Temperature
        default: 0
        number: 1
        on_read: |
          // 'address' contains the requested register address
          // 'value' contains the stored register value
          ESP_LOGD("ON_READ", "Temperature: address=%d, value=%d", address, value);

          float adc = id(temp_adc)->state;
          ESP_LOGD("ON_READ", "Temperature: temperature = %f", adc);

          int mv_int = (int)(adc * 1000000.0); // convert to integer mV
          ESP_LOGD("ON_READ", "Temperature:  temperature (mV) = %d", mv_int);

          return mv_int;
```

ayubn, 2023-08-23:
> This is my code. It doesn't update a switch directly, I called a script that does the update.
>
> There is an example for how to read a sensor at the bottom as well.
>
> ```
> uart:
>   id: modbus_uart [...]

hi, i am new to esphome

i hv two esp32 boards  .. first esp32 have connected some relay boards  through i2c.. and second esp32 have ld2410 presence sensor, ds18b20 temperature sensor and 1 IR transmitter

first board connected through lan card with home assistant and second board is connected through wifi they both working  fine under home assistant 

now i want to connect both boards through rs485... i want 2nd esp32 send all sensors data to 1st esp32 through rs485 and first device will connect to home asssistant via LAN8720

i want to know how to communicate both devices with rs485 (tried many codes but unable to connect them  properly ) for data read and  write ..... any sample code for both sides please

Kentrp, 2023-09-11:
> hi, i am new to esphome
>
> i hv two esp32 boards .. first esp32 have connected some relay boards through i2c.. and second esp32 have ld2410 presence sensor, ds18b20 temperature sensor and 1 IR [...]

Hi! Do you have decision, i need the same

wspeth, 2023-09-11:
I got it to work without major issues and ignoring any flow control. My
understanding is that flow control is just some timing behaviour handled by
the library.