epiclabs-uc/esphome-modbus-server #3: Wiring diagram

That-Dude, 2023-01-28:
First of all, thanks for your work, im excited to try it out. 

Am i right in thinking that you are connecting the modbus cables directly to your esp32 serial pins?

Or are you connecting those pins to a Max485 module or something similar?

A simple wiring diagram would be much appreciated.

Id like to connect esphome to my solar inverter which i think has -+ 5v for the modbus pins, i guess the esp32 will tolerate this but some clarification would be appreciated.

wspeth, 2023-03-04:
> First of all, thanks for your work, im excited to try it out.
> 
> How are you connecting the esp32 to modbus? A Max485 module or something similar?
> 
> A simple wiring diagram would be much appreciated.

Curious whether the same hardware was employes like 
https://esphome.io/components/modbus_controller.html

jpeletier, 2023-05-18:
Yes, the wiring is the same as in https://esphome.io/components/modbus_controller.html