epiclabs-uc/esphome-modbus-server #5: How to understand the guide?

StarChildDK, 2023-04-10:
This is in the guide:
holding_registers:
      - start_address: 79 # starting register range
        default: 82 # default value for all those registers
        number: 10 # number of registers in the range
        on_read: | # called whenever a register in the range is read
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGI("ON_READ", "This is a lambda. address=%d, value=%d", address, value);
          return value; // you can return the stored value or something else.

Let's say I want to return these holding registers:
Address 	Value
200  		398.413330   
202  		396.458252          
204  		396.623444   

How should I do that, like this?:

holding_registers:
      - start_address: 0 # starting register range
        default: 82 # default value for all those registers
        number: 18 # number of registers in the range
        on_read: | # called whenever a register in the range is read
          // 'address' contains the requested register address
          // 'value' contains the stored register value 
          ESP_LOGI("ON_READ", 200, 398.413330);
          ESP_LOGI("ON_READ", 202, 396.458252);
          ESP_LOGI("ON_READ", 204, 396.623444);
          return value; // you can return the stored value or something else.

Or how? I really hope you will specify for me.

StarChildDK, 2023-04-17:
Anyone?

StarChildDK, 2023-04-18:
Thank you so much!! 
Now I just have to figure out how to parse a HA value as an int. :)

jpeletier, 2023-05-18:
Thank you @That-Dude for your reply.

The component could be improved to simulate "float" registers, but that's a TODO.

closed 2023-05-18 by jpeletier